ArminJo/JK-BMSToPylontechCAN #7: Cell Info Page with maximum and minimum indicators

cfi700, 2023-08-06:
The indicator for the maximum is not always displayed correctly (see first row).
The indicator for the minimum does not appear at all.

[image]

Info: I use a battery with 16 cells and have set this in the source code accordingly.

ArminJo, 2023-08-06:
What is the Serial output telling?

cfi700, 2023-08-06:
I have captured some serial information.
The indicator looks always the same (Max: left top corner) (Min: not displayed)

SOC[%]=33 -> Remaining Capacity[Ah]=66
Battery Voltage[V]=52.09, Current[A]=-20.29, Power[W]=-1056

Total Runtime Minutes=41521 -> -> 28D20H 1M

*** CELL INFO ***
1=3257 mV, 2=3258 mV, 3=3258 mV, 4=3257 mV, 5=3256 mV, 6=3257 mV, 7=3257 mV, 8=3256 mV, 
9=3254 mV, 10=3256 mV, 11=3257 mV, 12=3254 mV, 13=3256 mV, 14=3256 mV, 15=3254 mV, 16=3256 mV, 
Minimum=3254 mV at cell #9, Maximum=3258 mV at cell #2 of 16
Delta=4 mV, Average=3256 mV<\r>

SOC[%]=33 -> Remaining Capacity[Ah]=66
Battery Voltage[V]=52.09, Current[A]=-20.68, Power[W]=-1077

commit 941681df (2023-08-06): Renamed cpp to hpp to fix inconsistent MAXIMUM_NUMBER_OF_CELLS values. closes #7
commit 9198c08e (2023-08-06): Renamed cpp to hpp to fix inconsistent MAXIMUM_NUMBER_OF_CELLS values. closes #7
closed 2023-08-06 by ArminJo