ArminJo/JK-BMSToPylontechCAN #13: Inverter read battery total capacity (norminal) wrong (50Ah) while serial log show different value (210Ah)

paulsteigel, 2023-10-22:
Hi Armin,
Today I tried a test with Arduino Uno R3 (ATMega328) at my parent's home and have the inverter connected successfully with the JK-BMS. However, Battery Capacity shown by Inverter was only 50Ah while the real value (read by Serial log was 210Ah - this is the correct number).
I doubt the CAN converter have a bit of problem on calculating. Can you please be so kind to help checking this with thanks.
Yours very faithfully,
Dang Dinh Ngoc

ArminJo, 2023-10-22:
> At long press, CAN Info page is entered and additional debug data is printed as long as button is pressed

What is the output of the debug page???

paulsteigel, 2023-10-22:
Hi Armin,
Thank you for your quick response. I connected a button to pin 3 (on A) and C to Ground (my button has 4 pin) - I currently dont have a 1K resitor to limit current - sorry. The following data is printed, I dont see CAN debuging data printed. I might be wrong then!

1= 0 % |    0,  2= 0 % |    0,  3= 0 % |    0,  4= 0 % |    0,  5= 0 % |    0,  6= 0 % |    0,  7= 0 % |    0,  8= 0 % |    0, 
 9=72 % |  150, 10= 0 % |    0, 11=27 % |   57, 12= 0 % |    0, 13= 0 % |    0, 14= 0 % |    0, 15= 0 % |    0, 16= 0 % |    0, 
Cell Maximum percentages
 1= 2 % |    7,  2= 0 % |    0,  3= 0 % |    0,  4= 0 % |    0,  5= 0 % |    0,  6= 0 % |    0,  7= 1 % |    6,  8=18 % |   59, 
 9= 0 % |    0, 10= 0 % |    0, 11= 0 % |    0, 12= 0 % |    0, 13= 2 % |    7, 14=15 % |   49, 15=16 % |   52, 16=43 % |  138, 

Warning: CellOvervoltageProtectionMillivolt value 3500 mV > 3450 mV is not recommended for LiFePO4 chemistry. There is less than 1% extra capacity above 3.5V.
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active

Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Battery Voltage[V]=52.44, Current[A]=-9.47, Power[W]=-496, Difference to full[V]=3.6
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Battery Voltage[V]=52.43, Current[A]=-9.60, Power[W]=-503, Difference to full[V]=3.6
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Battery Voltage[V]=52.43, Current[A]=-9.87, Power[W]=-517, Difference to full[V]=3.6
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Battery Voltage[V]=52.43, Current[A]=-10.00, Power[W]=-524, Difference to full[V]=3.6
Battery Voltage[V]=52.43, Current[A]=-9.60, Power[W]=-503, Difference to full[V]=3.6
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, not active
Battery Voltage[V]=52.43, Current[A]=-9.20, Power[W]=-482, Difference to full[V]=3.6
Charging MosFet enabled, active | Discharging MosFet enabled, active | Balancing enabled, active
Battery Voltage[V]=52.43, Current[A]=-9.47, Power[W]=-496, Difference to full[V]=3.6
Battery Voltage[V]=52.43, Current[A]=-9.74, Power[W]=-510, Difference to full[V]=3.6
Total Runtime Minutes=18920 ->   13D03H20M
* CELL INFO *
16 Cells, Minimum=3270 mV, Maximum=3280mV, Delta=10 mV, Average=3276 mV
 1=3279 mV,  2=3278 mV,  3=3278 mV,  4=3276 mV,  5=3278 mV,  6=3276 mV,  7=3279 mV,  8=3279 mV, 
 9=3271 mV,  10=3275 mV, 11=3270 mV, 12=3276 mV, 13=3278 mV, 14=3279 mV, 15=3279 mV, 16=3280 mV, 

* CELL STATISTICS *
Total balancing time=314 s ->   0D00H05M14S
Cell Minimum percentages

SOC[%]=32 -> Remaining Capacity[Ah]=67

ArminJo, 2023-10-22:
It must look like https://github.com/ArminJo/JK-BMSToPylontechCAN/blob/9f9be264ac7ef9d2e3c29954dffca267e3409461/extras/JK-BMS.log#L70

paulsteigel, 2023-10-22:
> It must look like
> 
> https://github.com/ArminJo/JK-BMSToPylontechCAN/blob/9f9be264ac7ef9d2e3c29954dffca267e3409461/extras/JK-BMS.log#L70

Hi Armin,
I updated the log file and discovered that CAN data frame does not include Total Capacity in Ah. That is why the Inverter does not reflect / and or calculate and just use its default value! (50Ah).
I also update the CAN log (previous comment) for your reference then!
Ngoc

ArminJo, 2023-10-22:
I must test it for backward compatibility...

paulsteigel, 2023-10-22:
Many thanks Armin,
It will be well appreciated by many users like my country then.
Also, I am thinking of adding a mod-bus with MAXRS485 to connect several batteries in parallel (the mater will connect with converter via CAN and slave batteries will connect together using mod-bus.
Thanks again for all your supports!
Ngoc

ArminJo, 2023-10-23:
Hi Ngoc,
what is the sensible content of the 6 non capacity bytes in frame 0x35F? I guessed some values.

What is the sensible content of `SOCHighDefinition100PPM` as 3. word of frame 0x355, documented in SMA CAN protocol. I did not include it so far.

There should be no difference between the 328 versions, except if this are counterfeit versions.
My two China clone Nanos are working fine.

paulsteigel, 2023-10-24:
> Hi Ngoc, what is the sensible content of the 6 non capacity bytes in frame 0x35F? I guessed some values.
>
> What is the sensible content of `SOCHighDefinition100PPM` as 3. word of frame 0x355, [...]

Hi Armin,
Regarding this table, I assume only the 4th and 5th are meaningful for the capacity in Ah to be sent to the inverter. 
[image]

I am now checking a bit more with other type of batteries to see whether these can be meaningful and worth incorporating into the PylontechCAN. At the moments, the SOC is enough for the inverter to control charging/discharging.

[image]
What I am wondering is how the inverter can have this data for some other type of the batteries (SVE5000WM), it can read the Capacity in Ah and also apply a nice charging curve for the battery (see second image). So it means that the Inverter can request more data than we have included in the CAN message and probably have its internal charging computation so far. 
Best,
Ngoc

closed 2023-10-24 by ArminJo