ArminJo/JK-BMSToPylontechCAN #29: Code Signal Flag from inverter to indicate CAN comms is working?

rooftopsolarsa, 2023-12-16:
Hey ArminJo :)

I am trying to add status LED's to show if COMMS is working or not.

On the JKBMMS side I have managed to adjust/add to the code to allow LED to be on/off depending on status of BMS comms working or not.

I have defined STATUS LED's for CAN and BMS and POWER status
Then..
AT LINE 698 in JK-BMSToPylontechCAN.ino

```
bool readJK_BMSStatusFrame() {
    sMillisOfLastReceivedByte = millis();
    uint8_t tReceiveResultCode = readJK_BMSStatusFrameByte();
    if (tReceiveResultCode == JK_BMS_RECEIVE_FINISHED) {
        /*
         * All JK-BMS status frame data received
         */

// START MY CODE HERE TO SWITCH ON LED
digitalWrite(BMS_STATUS_LED_PIN, HIGH); // Turn on LED IF BMS CONNECTED and Receiving - WORKING
// END MY CODE

        return true;

    } else if (tReceiveResultCode != JK_BMS_RECEIVE_OK) {
        /*
         * Error here
         */

// START MY CODE HERE TO SWITCH OFF LED
digitalWrite(BMS_STATUS_LED_PIN, LOW); // Turn on LED IF BMS CONNECTED and Receiving - WORKING
// END MY CODE HERE TO SWITCH OFF LED

        Serial.print(F("Receive error="));
        Serial.print(tReceiveResultCode);
        Serial.print(F(" at index"));
        Serial.println(sReplyFrameBufferIndex);
        sFrameIsRequested = false; // do not try to receive more
        sBMSFrameProcessingComplete = true;
        printJKReplyFrameBuffer();
    }
    return false;
}
```

Now I want to do the same for the CAN side.
In the code I can find where to see status if CAN Module is on or not. But I cannot find any code which shows any signal/flag/callback coming from the inverter to say all comms are working.

Currently it looks like the CAN is one-way only, sending to the inverter, and nothing coming back from the inverter?
Or advise suggest another way to know in the code if CAN is communicating or not?

ArminJo, 2023-12-17:
Yes you are right. The CAN communication is one way. I do not try to receive any response or acknowledge from the inverter, but maybe there is a response or acknowledge I am not aware of.

rooftopsolarsa, 2023-12-17:
Ok great, I thought I was going crazy to not find anything coming back from inverter.
So it will have to be enough to know the CAN MODULE is working by showing LED on.
Now I can maybe just flash the led each time CAN is sent to inverter.

In your code at line 546
```
    /*
     * Send CAN frame independently of the period of JK-BMS data requests
     * 0.5 MB/s
     * Inverter reply every second: 0x305: 00-00-00-00-00-00-00-00
     * Do not send, if BMS is starting up, the 0% SOC during this time will trigger a deye error beep.
     */
    if (sCANDataIsInitialized && !JKComputedData.BMSIsStarting
            && millis() - sMillisOfLastCANFrameSent >= MILLISECONDS_BETWEEN_CAN_FRAME_SEND) {
        sMillisOfLastCANFrameSent = millis();

          digitalWrite(CAN_STATUS_LED_PIN, LOW); // Turn OFF LED IF CAN SENT
          delay(20);
          digitalWrite(CAN_STATUS_LED_PIN, HIGH); // Turn ON LED BACK ON

        if (sDebugModeActivated) {
            Serial.println(F("Send CAN"));
        }
        sendPylontechAllCANFrames(sDebugModeActivated);
    }
```
I am just not sure how the delay(n) will affect the timing in your code?
Or do you have a better more elegant solution to flash the LED each time CAN is sent to the inverter?

ArminJo, 2023-12-17:
Thanks for the inspiration 🥇 
I added code for communication status LEDs and tested it.

closed 2023-12-17 by ArminJo