ArminJo/JK-BMSToPylontechCAN #36: UART received from JKBMS(B2A8S20P) needs offset to ground in order for Arduino Nano 5V to hear it.

agmkv70, 2024-08-23:
Hello, after setting up recommended design, I found out that Nano(5V 328p) wasn't hearing the BMS. It occurred that having VCC at 5.09V (when line at 1) and 5.09-2V at sending 0 was not enough to trigger RX to be heard! (2V - amplitude of BMS signal on oscilloscope)
Well, so I put a random pulldown 0.6kOhm resistor at Nano's RX and it worked just fine. Could be of use for someone else.

ArminJo, 2024-08-23:
Thanks for the hint 👍 
Are you using a shottky diode in the BMS TX line, or is it a plain silizium diode?
You can mesure it with the diode range of a multimeter, the shottky diode should have 0,2 to 0.3 V.

agmkv70, 2024-08-23:
I do not use any diode now (programming needs taking the board away from the room with bms), but I tried putting in some 0.59V diode. voltage seemed to drop by about 0.5 and it still wasn't enough

ArminJo, 2024-08-23:
No diode is optimal, with a 0.59V (SI) diode it will most probably not work.
600 Ohm seems to be a bit low, can you check it with 1 or 2 k Ohm?
Depending on your results, I will document the pull down resistor in the readme then.
Thanks

agmkv70, 2024-08-23:
OK, I'll check in the evening or tomorrow

agmkv70, 2024-08-23:
I have a nice robotdyn version of nano like this: https://www.cgtrader.com/3d-models/electronics/other/arduino-nano-v3-with-microchip-atmega328p-from-robotdyn
on CH340G.
Resistor is marked as 680 Ohm, and has 673 Ohm.
Signal range on RX is just fine 3.63V max and amplitude:1.65V. So I guess no reason to even try anything else...

closed 2024-08-23 by ArminJo